Jeija/bluefluff #1: Windows

gdm29732, 2017-01-15:
I don't see any instructions for Windows.  Does this only work for Linux and Mac

Jeija, 2017-01-15:
This should also work on Windows, but I haven't tested that yet. However, it is a lot more complicated to install, since Windows doesn't have a nice command line. Someone has made a tutorial on how to install noble on Windows, which is the only thing platform-dependent that this needs:
https://www.youtube.com/watch?v=mL9B8wuEdms
Once noble and the other dependencies are installed, running fluffd should basically work just like on Linux / macOS.

Someone else has suggested it might actually be easier to install Linux and then install fluffd on Linux: http://www.tamatalk.com/IB/topic/194076-furby-connect-bluetooth-hacking/

gdm29732, 2017-01-15:
> topic/194076-furby-connect-bluetooth-hacking/

I am using windows.  I have noble installed.  When I do node fluffd.js It
discovers furby, but I get a type error saying that Object.values is not a
function. I ave pasted it below.

noble warning: unknown handle 64 disconnected!
info: Discovered Furby: f264da436ac1
C:\bluefluff\fluffd\fluffcon.js:234
                let characteristicUUIDs =
Object.values(FURBY.CHARACTERISTIC);
                                                 ^

TypeError: Object.values is not a function
    at C:\bluefluff\fluffd\fluffcon.js:234:36
    at Peripheral.<anonymous>
(C:\bluefluff\fluffd\node_modules\noble\lib\peripheral.js:38:7)
    at Peripheral.g (events.js:291:16)
    at emitOne (events.js:96:13)
    at Peripheral.emit (events.js:188:7)
    at Noble.onConnect
(C:\bluefluff\fluffd\node_modules\noble\lib\noble.js:148:16)
    at emitTwo (events.js:106:13)
    at NobleBindings.emit (events.js:191:7)
    at NobleBindings.onLeConnComplete
(C:\bluefluff\fluffd\node_modules\noble\lib\hci-socket\bindings.js:212:8)
    at emitMany (events.js:127:13)
I looked at the instructions for linux, but that will not work on my
computer.  I have nrf connect on my phone.  Is there a way to use that to
send commands other than the db command.  I tried to enter some of the
other commands, but could not get them to work.

Thanks

Jeija, 2017-01-15:
Yes, you could also send other commands from the nRF connect app, e.g. "cd00" will turn off the LCD eyes ("cd01" to turn them back on again), but sending longer commands becomes really tedious on the phone.

Now concerning the error:
I'm not sure what `noble warning: unknown handle 64 disconnected!` means here, but since `info: Discovered Furby: f264da436ac1` is displayed afterwards, I would assume everything is fine. This means your Bluetooth and everything is working, you're just one step away from controlling the Furby.

Now the actual cause for your problem is that I used some very new node.js (ECMAScript 2017) features when writing fluffd, that is why you need the current node.js version (you propably got the LTS version, which doesn't seem to work). I have tested this with version 7.4, so that should work. You can get the current node.js from [this page](https://nodejs.org/en/download/current/). Let me know if that helps you!

gdm29732, 2017-01-16:
Hello,

I was able to get node fluffs.js to start but all it said in the CMD line
is lines of verbose GP notification and verbose general plus write.  This
stopped and there is now just a blinking line.  I opened the html and it
says no parameters

Jeija, 2017-01-16:
If you get those GP notification and GeneralPlusWrite lines, then everything is working fine (e.g. the GP notifications should come up when you move Furby's antenna). You only need to make sure Furby stays connected (Furby should keep quiet while connected) and open the HTML while the server is running.

If that doesn't work, you should check out your browser's JavaScript error console to see what the issue might be. It could also help to serve the HTML on a local HTTP server.

gdm29732, 2017-01-16:
I got it to work.  There was a problem with my debugger settings in the
browser.  Thank you so much!  This program is awesome!

Jeija, 2017-01-16:
Good to hear that 😄 

Closing this issue now.

closed 2017-01-16 by Jeija