Jeija/bluefluff #2: fluffcon.js Errors?

saintfelony, 2017-01-19:
I'm fairly new to the whole Bluetooth end of things but I think I have Node/Noble set it right since the test and examples in the Noble install seem to be working. 

On Windows 10 when I go to run fluffd I get: 

D:\Furby\bluefluff-master\fluffd\fluffcon.js:234
                let characteristicUUIDs = Object.values(FURBY.CHARACTERISTIC);
                                                                    ^

TypeError: Object.values is not a function
    at D:\Furby\bluefluff-master\fluffd\fluffcon.js:234:36
    at Peripheral.<anonymous> (D:\Furby\bluefluff-master\fluffd\node_modules\noble\lib\peripheral.js:38:7)
    at Peripheral.g (events.js:291:16)
    at emitOne (events.js:96:13)
    at Peripheral.emit (events.js:188:7)
    at Noble.onConnect (D:\Furby\bluefluff-master\fluffd\node_modules\noble\lib\noble.js:148:16)
    at emitTwo (events.js:106:13)
    at NobleBindings.emit (events.js:191:7)
    at NobleBindings.onLeConnComplete (D:\Furby\bluefluff-master\fluffd\node_modules\noble\lib\hci-socket\bindings.js:21
2:8)
    at emitMany (events.js:127:13)

Any ideas or is this something on my end?

Jeija, 2017-01-19:
You have to update your node.js to v7.0 or later, I am using v7.4.0. Also see #1 where we had the same discussion. You can get the latest nodejs from https://nodejs.org/en/.

saintfelony, 2017-01-19:
Oops. The closed issue tab is not very obvious, however that doesn't exactly cover what I'm running into. 

I installed 7.4.0 initially and could not get Noble past a USB module error on it compiling. I switched to 6.9.4LTS and Noble worked just fine. I'll have to dig into Noble some more and see where the problem lies, or I'm wondering if I can compile Noble in 6.9.4 and then update to 7.4.0 and have Fluffd work. 

Hopefully I can get Noble to cooperate and poke at this some too. This looks like a lot of fun.

saintfelony, 2017-01-19:
I spent a good part of the afternoon poking at it and am having no luck with node.js 7.4.0 and Win10. Both Noble and Fluffd fail on install due to missing modules and a few other errors. I got past the USB module problem if I turn of any AV but then still has debug issues. Node.js 6.9.4lts outside of a few warnings works for Noble and just fine outside of the specific new calls you're using for Fluffd.  

I still haven't tried installing Noble, updating and then installing Fluffd but I'm not having high hopes of that working because I'm assuming it does some version checking. 

I'm am curious to see if I can get it working since it sounds like some people did already but it may just work better to set up a VM with linux...

jcostlow, 2017-04-02:
This patch will apply cleanly. I'm not doing a pull request because values is better.

```diff
diff --git a/fluffd/fluffcon.js b/fluffd/fluffcon.js
index f32f6e0..8f675ac 100644
--- a/fluffd/fluffcon.js
+++ b/fluffd/fluffcon.js
@@ -231,7 +231,8 @@ module.exports.connect = function(furby, callback) {

                exitHandler(furby);

-               let characteristicUUIDs = Object.values(FURBY.CHARACTERISTIC);
+               //let characteristicUUIDs = Object.values(FURBY.CHARACTERISTIC);
+               let characteristicUUIDs = Object.keys(FURBY.CHARACTERISTIC).map(function(key) {return FURBY.CHARACTERISTIC[key];});
                getFurbyCharacteristics(furby, FURBY.SERVICE.FLUFF, characteristicUUIDs, function(characteristics) {
                        let gpWrite = characteristics[FURBY.CHARACTERISTIC.GENERALPLUS_WRITE];
                        let gpListen = characteristics[FURBY.CHARACTERISTIC.GENERALPLUS_LISTEN];
```