Jeija/bluefluff #3: Missing Name - Theory

bshep, 2017-01-22:
> Even though the name is in the list in the Furby Connect World App and even though it should exist in the regular pattern of possible two-syllable Furby names, you cannot name your Furby Connect "Tay-Tah".

I suspect this is due to profanity filtering.  "Tay-Tah" when sounded out in spanish would be "Teta" which is similar to saying "boob" in spanish.

Jeija, 2017-01-22:
That would make a lot of sense, this is the kind of explanation I was suspecting. Spanish is widely spoken enough so that this might have actually impacted Hasbro's decisions. Thanks for sharing!

ja450n, 2018-01-13:
Interestingly enough, if you send action 33,0,0,106 furby will say Tay-Tah