Jeija/bluefluff #5: furby often disconnects. fluffd shows noble warning unknown peripheral.

dvvincent, 2017-01-26:
first i'd like to say thanks for putting this up on github, i purchased two furbies before christmas just for fooling around with the intention of completing some bluetooth sniffing. you beat me to it!.

on to the question.. i  noticed that often that fluffd will lose connection to the furby and when i review the console, i see noble warning: unknown peripheral myfurby UUID. the funny thing is that it's the same furby so why does noble find it the second time?

also when this occurs, often you can drop out of fluffd, using ctrl-c states "closing connection" but stalls..

any ideas?

dvvincent, 2017-01-26:
on second thought i'm gonna put in a fresh set of batteries.. my other furby doesn't this issue as much.

Jeija, 2017-01-26:
Some of the suggestions I give when Furby doesn't seem to connect are
* Use fresh batteries or a lab power supply, don't use rechargeables!
* Place Furby closer to the BLE adaptor (~50cm works for me)
* Maybe try a different BLE adaptor (I am using [this one](https://www.amazon.com/Rocketek-Low-Energy-Bluetooth-Adapter/dp/B00H8O8CMO))

I hope fresh batteries fixed the issue for you! By the way, contributions are weclome - so if you make some discoveries concerning Furby's BLE protocol / the hardware / DLC files / actions, a pull request would be great!

Jeija, 2017-01-26:
Also one very specific thing: One key piece that is still missing is how to trigger the "hot chilli animation" [(the one in this video)](https://youtu.be/9GEAIgOgy_o?t=56). Once we find that out, we should be able to display arbitrary content in Furby's LCD eyes, so any information on how to play that sequence is greatly appreciated.

dvvincent, 2017-01-26:
i'll try fresh batteries tonight.. as for the contributions, i've been working on testing out all the actions starting at 1 and working my way up. the most difficult part is trying to translate what the furby is actually saying! 

also there doesn't seem to be any order to why certain actions have multiple specifics, some all the way to 7 or 8, while some end much earlier. like you mentioned earlier, the actions seems to be collected together by emotion, etc.

Jeija, 2017-01-26:
Some more documentation on all the actions would be great 👍 !

If that works for you, I'd prefer collecting the information in a machine-readable format so that it can later also be used with fluffd (e.g. allow searching for actions based on what you want Furby to say). I have made a suggestion for a possible format [over here](https://github.com/Jeija/bluefluff/pull/4#issuecomment-275206099), but if you prefer a different structure, that would also work for me. Maybe we also need some more properties, e.g. the "emotion" of the action or maybe we should name the different subindices / indices / inputs (e.g. 74.0.1.* could be called "princess").

dvvincent, 2017-01-26:
JSON looks to be the way to go. your example looks like a reasonable place to start. i like idea of including section for eyes, etc.

dvvincent, 2017-01-27:
if anyone is looking for internal pictures without ripping it apart, the fcc has done all the work for you.

https://fccid.io/RS4-B6083A

just scroll down to the link for internal photos. sadly the processors are capped with epoxy blobs..

DigitalizedWarfare, 2017-01-28:
What node version needs to be installed?  or java? Im getting this. I have order the adapter you have used, waiting for it to get here, using a Gigaware bluetooth 4.0.
Node version i am using is v6.9.4

c:\bluefluff\fluffd>node fluffd.js
noble warning: unknown handle 73 disconnected!
info: Discovered Furby: fe8ea12cb20d
c:\bluefluff\fluffd\fluffcon.js:234
                let characteristicUUIDs = Object.values(FURBY.CHARACTERISTIC);
                                                 ^

TypeError: Object.values is not a function
    at c:\bluefluff\fluffd\fluffcon.js:234:36
    at Peripheral.<anonymous> (c:\bluefluff\fluffd\node_modules\noble\lib\peripheral.js:38:7)
    at Peripheral.g (events.js:291:16)
    at emitOne (events.js:96:13)
    at Peripheral.emit (events.js:188:7)
    at Noble.onConnect (c:\bluefluff\fluffd\node_modules\noble\lib\noble.js:148:16)
    at emitTwo (events.js:106:13)
    at NobleBindings.emit (events.js:191:7)
    at NobleBindings.onLeConnComplete (c:\bluefluff\fluffd\node_modules\noble\lib\hci-socket\bindings.js:212:8)
    at emitMany (events.js:127:13)

Jeija, 2017-01-28:
Since this has come up several times now, I have now [also added a hint in the README](https://github.com/Jeija/bluefluff/commit/fd6ff7381d225ac8a8e7fbb6cb3a786271a6212e). *You will need to update your node.js to version 7.0 or later!* It might technically also work to use node.js 6.* and launch fluffd using `node --harmony fluffd.js`, but that is unsupported by the node.js developers.

DigitalizedWarfare, 2017-01-28:
ok cool. Adding the flag --harmony worked great for node 6.9 Thanks! Now to find the dlc and mod them.

DigitalizedWarfare, 2017-01-28:
Can you tell us how to create a dlc? or what software to view, edit, or create a dlc?  This project is great!

Jeija, 2017-01-28:
It's not yet possible to create your own DLC from scratch, I haven't got that far yet. I think [this](https://github.com/Jeija/bluefluff/blob/master/doc/connectworld.md) should answer where to get official DLC files from, [this](https://github.com/Jeija/bluefluff/blob/master/doc/dlcformat.md) should give you a rough idea on how to visualize and edit DLCs and [this](https://github.com/Jeija/bluefluff/blob/master/doc/flashdlc.md) tells you how to flash your own DLCs.

DigitalizedWarfare, 2017-01-30:
Yup i read all of those. I was just hoping you had more information, but thanks man!

closed 2017-04-03 by Jeija