Jeija/bluefluff #7: Unable to send dlc

DigitalizedWarfare, 2017-02-01:
I tried every node version after 6.9. Up to like 7.4. Anything above 6.9 files on windows 10 64.  So go to 6.9 run with the harmony switch and it will load. When try to send the dlc to it, the app crashed and throws an error. No error log is generated. I purchased the same adapter as you said you used, also bought two off the usb site support list. each fails Other versions of node fail to build. node cant download files for usb, etc.. MSbuild failes.. Here is the output with node 6.9 Any ideas please?
 
c:\bluefluff\fluffd>node --harmony fluffd.js
info: Discovered Furby: fe8ea12cb20d
info: Connected to Furby
debug: Read all fluff characteristics
verbose: generalPlusWrite: 00
verbose: GP notification: 2100000001000000
verbose: GP notification: 2100010001000000
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: Sending flashdlc command to all Furbies, params: filename=custom.dlc, dlcfile=
warn: Could not parse HTTP command: Error: ENOENT: no such file or directory, stat ''
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: Sending flashdlc command to all Furbies, params: filename=custom.dlc, dlcfile=C:\bluefluff\fluffd
verbose: generalPlusWrite: 500000000002637573746f6d2e646c630000
verbose: GP notification: 2402
info: FlashDLC: Got Ready to Receive
c:\bluefluff\fluffd\fluffaction.js:162
                                        throw err;
                                              ^

ReferenceError: err is not defined
    at c:\bluefluff\fluffd\fluffaction.js:162:12

c:\bluefluff\fluffd>

DigitalizedWarfare, 2017-02-01:
Also changing the line from throw err; to 
if (error) {
	winston.log('There was an error', error);
  return;
}

allows the app to continue instead of crashing. So i tried to send the stock dlc to it and this is what i get. 

[image]

DigitalizedWarfare, 2017-02-02:
In addition, i upgraded to node 7 on my kali box and this error still pops on trying to flash the dlc to the furby. I edit the file for the error as above and sent the dlc. Its been around 15 to 20 minutes and all i am seeing is the generalpluswrite : 00 line and nothing else. it never says 2403 and in the debug it never changes from 0. so it looks like its not sending the dlc to the furby. Again i am using the exact usb bluetooth you are saying.

LOL. i feel like im almost there. I just need to figure out this last step. I can do everything else including delete slots. Just not write it seems.

DigitalizedWarfare, 2017-02-02:
I kind of have it going, but the furby keeps failing asleep while flashing.. lol....

Jeija, 2017-02-02:
The errors you were mentioning sound like fluffd is unable to read the DLC file on your filesystem. Make sure the `dlcfile` path you specify is correct and the user fluffd is running as has read permissions. There also seems to be a bug in fluffd where you have to restart fluffd each time before flashing a DLC, so you could either patch that bug (the flashing callback should not be attached multiple times) or just restart fluffd every time.

furbytamer, 2017-03-12:
Is there a way to set up my furby for all this with an s4? I don't have a personal computer or apple product.

closed 2017-04-03 by Jeija