Jeija/bluefluff #8: Android support?

furbytamer, 2017-03-12:
I don't have a computer or apple device how do i hack my furby from my s4?

Jeija, 2017-03-12:
You can manually send the commands specified [here](https://github.com/Jeija/bluefluff/blob/master/doc/generalplus.md) to the Bluetooth characteristic with ID `dab91383b5a1e29cb041bcd562613bde` using the nRF connect app (as demonstrated in my YouTube video). That way you should be able to enter debug mode (send `db`), change Furby's mood (send e.g. `24010010` to set excitedness to `0x10` = 16), control the LCD backlight (send `cd00` or `cd01`) or execute preprogrammed sound file + movement sequences (send e.g. `13004a0001` to make him say something about being a DJ).

You won't be able to upload your own audio to the Furby. This process is fairly complicated and involves a lot more than just sending the correct commands (e.g. you need to convert the audio file to the A1800 codec and align the audio inside the DLC), so you will certainly need a computer for that.

furbytamer, 2017-03-12:
Will there be moble apps in the future for this?

Jeija, 2017-03-12:
I'm not planning to make one and I don't know about anyone else creating a mobile app for this, so I'm afraid there are not going to be any mobile apps, not even mentioning the legal issues creating a mobile version might cause.

furbytamer, 2017-03-12:
How do you restore the eyes to normal?

Jeija, 2017-03-12:
Keep sending `db` commands until they get back to normal or take out the batteries.

furbytamer, 2017-03-12:
Is there any commands for changing the personality value?

Jeija, 2017-03-12:
No, I've been speculating that Hasbro is going to sell additional masks (like the sleep mask) for Furby Connect that turn Furby into Ninja, Princess, DJ etc. However, I've also heard reports that the audio / eye animations for these personalities are not stored on the mask chips themselves, but are already inside Furby's memory. But I don't know how to access these personalities just yet.

furbytamer, 2017-03-12:
Im not much of a code expert but if i find anything cool ill let you know :3

furbytamer, 2017-03-12:
I remember the antenna was rumored to function as a joystick  so maybe ill try connecting it to an emulator :3

furbytamer, 2017-03-12:
His excitement, tired, and displeasure values seem to increase in response to moving the antenna

aptitudex, 2017-03-14:
Does the antenna have any functional use inside the app? Does it actually do anything other than look pretty?
I don't have the Furby Connect yet myself, so I'm going out on a bit of a limb here.

Jeija, 2017-03-14:
It didn't seem to have any use inside the app last time I checked, but I would assume this is a feature that might be added in the future. The `0x21` GeneralPlus response contains some very detailed information on antenna position (can detect all directions, including when the antenna is pushed down), so technically the antenna could be used as a joystick.

And of course it functions as an indicator for Furby's various "desires", e.g. sometimes it glows blue and Furby says something along the lines of "something happening at the egg machine".

ApertureAndroid, 2017-03-14:
When you're connected to the app, and you wiggle/press the antenna, Furby will say various things. When not connected, all it does is change color.

@Jeija, I couldn't figure this out myself; What value do you send using NRF Connect to have Furby start singing one of his DLC'd songs?

Jeija, 2017-03-14:
Oh wow, I never got that "Downloading broccoli schematics", interesting to hear Furby does that 😏 

@ApertureAndroid: The DLC sounds are mapped to Input 75 which is hex `0x4B`, so the command for launching one of these actions should be something like `13004b000001`. You propably want to play around with the last 3 bytes (6 characters) which are "index", "subindex" and "specific" of the action. Depending on the current DLC, these correspond to different songs / reactions / song announcements.

ApertureAndroid, 2017-03-14:
@Jeija Oh, I see. I was trying to use the number "75." I didn't think to use the hex code. Interesting; the example you gave seems to run some speech used in a DLC that's not even accessible in the app currently...

closed 2017-04-03 by Jeija

mew15845, 2018-09-16:
I think hasbro has completely forgotten that the poor furbies even exist! If you look at their update history in the app store they've been inactive for a year now. Is there an SD card in the furbies? 
I love my furbie but i'd like to make it sound less obnoxious and gassy as well as actually respond to vocal commands.

snakeroot1978, 2019-01-05:
Hi,

I bought a furby connect for a little 6 year old girl. It doesn't move not even its eyes are there any nrf codes that i can try to get it to move ? Please help when she was very young she was very ill too.