Jeija/bluefluff #9: May-Kah and May-Tah's values are swapped in the "names" list.

ApertureAndroid, 2017-03-16:
I wanted to name my Furby "May-Kah," so I sent "130021000056" using the NRF connect app, but it ended up making it "May-Tah." So I tried "130021000058" and got it to "May-Kah."

Jeija, 2017-03-16:
I can't reproduce this, for me May-Kah is `0x56` and May-Tah is `0x58`. Sometimes the pronounciation is not that clear. Could you try other first syllables to check if maybe you only got confused by Furby's pronounciation of `Kah` vs `Tah`?

ApertureAndroid, 2017-03-16:
I tried many times, just to check if I was mishearing, but it's a lot more clear when I have it greet my other Furby. I'll try and grab a video or something.

ApertureAndroid, 2017-03-17:
Okay, I don't know what changed, but the names match up now. Maybe I _was_ hearing it wrong. Whoops.

closed 2017-03-17 by ApertureAndroid