Jeija/bluefluff #15: error: Error while conencting: Error: Connection Rejected due to Limited Resources (0xd)

yabmtm, 2017-08-22:
When running sudo node fluffd.js, I do see a discovered furby, but then it immediately puts the error message:

error: Error while conencting: Error: Connection Rejected due to Limited Resources (0xd)

then it hangs.
I'm using node 8.4.0.

Jeija, 2017-08-23:
I've never had this issue, but this sounds like something that is caused by your Bluetooth adapter. In the [noble README](https://github.com/sandeepmistry/noble) this error is specifically mentioned for the Intel Dual Band Wireless-AC 7260 card, but it might affect other BLE adapters as well. Are you using the Intel AC 7260 by chance?
Another option might be that some other process on your system also tries to access the BLE adapter at the same time, but that sounds unlikely here. You could try to use a USB Bluetooth dongle instead (I've been using one of [these](https://www.amazon.com/Rocketek-Low-Energy-Bluetooth-Adapter/dp/B00H8O8CMO)).

yabmtm, 2017-08-27:
Yup, I have the 7260. I'll order one of the usb dongles and give that a shot. Thanks!

yabmtm, 2017-11-03:
Sorry, forgot to comment again. The USB dongle worked great, thanks!

closed 2017-11-03 by yabmtm