Jeija/bluefluff #20: Cannot connect to Furby on MacOSX

Bartman236, 2017-10-28:
I installed node 8.8.1 and this seems to go well. Only a few warnings, no errors:

npm WARN fluffd@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: brace-expansion@1.1.6 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/minimatch/node_modules/brace-expansion):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: concat-map@0.0.1 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/minimatch/node_modules/brace-expansion/node_modules/concat-map):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: balanced-match@0.4.2 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/minimatch/node_modules/brace-expansion/node_modules/balanced-match):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: inherits@2.0.1 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/inherits):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: wrappy@1.0.2 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/inflight/node_modules/wrappy):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fs.realpath@1.0.0 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/fs.realpath):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: rimraf@2.5.4 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: request@2.74.0 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: tunnel-agent@0.4.3 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request/node_modules/tunnel-agent):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: tough-cookie@2.3.1 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request/node_modules/tough-cookie):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: stringstream@0.0.5 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request/node_modules/stringstream):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: qs@6.2.1 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request/node_modules/qs):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined

added 152 packages in 6.204s

However, when I try to connect to Furby by starting "node fluffd.js" nothing seems to happen. I can open the HTTP client and send commands, but no response from Furby. What does work is connecting to Furby via NRF connect on my iPhone and getting it into debug mode via the "db" command. The Furby app also fully works.
What am I doing wrong? It cannot be my Macbook hardware because it late 2013 and therefore supports BLE.

Jeija, 2017-10-28:
Are you using macOS High Sierra by chance? According to [this noble library issue](https://github.com/sandeepmistry/noble/issues/679), the BLE bindings currently seem to be broken. Also, if I recall correctly you might need to install the `xpc-connection` npm package for noble to work properly.

Bartman236, 2017-10-28:
Indeed I am. I tried to install xmc-connection and it gives me the following errors:

gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/Users/bart/Downloads/bluefluff-master/fluffd/node_modules/xpc-connection/build'
gyp ERR! System Darwin 17.0.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/bart/Downloads/bluefluff-master/fluffd/node_modules/xpc-connection
gyp ERR! node -v v8.8.1
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN fluffd@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: inflight@1.0.5 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/inflight):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: once@1.3.3 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/once):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: wrappy@1.0.2 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/once/node_modules/wrappy):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: minimatch@3.0.3 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/minimatch):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: brace-expansion@1.1.6 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/minimatch/node_modules/brace-expansion):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: concat-map@0.0.1 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/minimatch/node_modules/brace-expansion/node_modules/concat-map):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: balanced-match@0.4.2 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/minimatch/node_modules/brace-expansion/node_modules/balanced-match):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: inherits@2.0.1 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/inherits):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: wrappy@1.0.2 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/inflight/node_modules/wrappy):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fs.realpath@1.0.0 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf/node_modules/glob/node_modules/fs.realpath):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: request@2.74.0 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: tunnel-agent@0.4.3 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request/node_modules/tunnel-agent):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: semver@5.3.0 (node_modules/usb/node_modules/node-pre-gyp/node_modules/semver):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: rimraf@2.5.4 (node_modules/usb/node_modules/node-pre-gyp/node_modules/rimraf):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: tough-cookie@2.3.1 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request/node_modules/tough-cookie):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: stringstream@0.0.5 (node_modules/usb/node_modules/node-pre-gyp/node_modules/request/node_modules/stringstream):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Cannot read property 'type' of undefined

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! xpc-connection@0.1.4 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the xpc-connection@0.1.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/bart/.npm/_logs/2017-10-28T20_34_58_541Z-debug.log

Jeija, 2017-10-29:
Well, if you're using High Sierra there is nothing you can do now but wait until the noble developers add High Sierra support (or try bluefluff on another OS). Since there already is a pull request for noble, that should just be a matter of days / weeks. And for installing `xpc-connection`: You probably need `sudo` before `npm install`.

tetreault, 2019-06-20:
for continuity's sake im just reporting i'm having similar (but not 100% similar) errors on MacOS Mojave

tetreault, 2019-06-21:
sorry to bump an old issues @Jeija but I just got fluffd working on MacOS Mojave 10.14.5 by switching from `noble` to `@abandonware/noble`

tetreault, 2019-06-21:
```shell
Stephens-MacBook-Pro-2:fluffd stephentetreault$ node fluffd.js
info: Discovered Furby: f5fc63e59e024d81a6be371983bd6ae8
info: Connected to Furby
debug: Read all fluff characteristics
(node:18814) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
verbose: generalPlusWrite: 00
verbose: GP notification: 2100000001000000
verbose: GP notification: 200e
verbose: GP notification: 2100030001000000
verbose: GP notification: 2100020001000000
verbose: GP notification: 2100000001000000
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
verbose: generalPlusWrite: 00
```

hartos, 2019-07-19:
Thanks @tetreault - switching from `noble` to `@abandonware/noble` worked for me too.

**Changes:**

file::package.json

in the section  `"dependencies": {`
change
		 `"noble": "1.9.1",`
to
		`"@abandonware/noble": "^1.9.2-2",`
and

file::fluffd.js
change
		 `const noble = require("noble"); `
to
		 `const noble = require("@abandonware/noble");`

Great project - thank you  @Jeija

PwRhGaStPY, 2020-05-23:
> Well, if you're using High Sierra there is nothing you can do now but wait until the noble developers add High Sierra support (or try bluefluff on another OS). Since there already is a pull request [...]

do you mean this as in `sudo npm install` or something else?