Jeija/bluefluff #23: Error while connecting

olhor, 2017-12-06:
This what happens after running `node fluffd.js`
`
info: Discovered Furby: <furby_mac>
error: Error while connecting: Error: Connection Rejected due to Limited Resources (0xd)
`
Any ideas?

Jeija, 2017-12-07:
See https://github.com/Jeija/bluefluff/issues/15
If you have the Intel AC 7260 wireless card, that should explain your problem.

olhor, 2017-12-07:
Thanks for quick response. I do not know how did I miss it from 'noble' docs... I checked them on how to write sample test app. 
If this can solve the problem:

> You need to stop scanning before trying to connect in order to solve this issue.

and I am able to get my BT adapter info from 'noble' I will try to make a PR to the repo fixing the issue.

closed 2017-12-08 by olhor