Jeija/bluefluff #26: Fluffd not working on PC

XxDJPartyGirl1034xX, 2018-01-03:
Hey, Jeija, i must say, i have learned how to make my Furby connect do things, like say phrases for future app gameplay, (Shower, games, etc). BUT... the thing i rlly wanna do is put my custom images on the Furby's LCD eyes, and custom audio, but all i have is NRF connect on IOS. I would like to know if you could flash your custom audio and images to the eyes. Any help? Thank you!

Jeija, 2018-01-03:
No, I'm afraid you won't be able to do that just using NRF connect, at least there is no way I'm aware of. I'd really like to make it easier for people to upload their custom audio files by e.g. making an Android app or something, but I don't have the time and I also don't want to get into legal issues.

There have been others talking about difficulties to install NRF connect, maybe this Issue post can help you: https://github.com/Jeija/bluefluff/issues/1#issuecomment-272678385

XxDJPartyGirl1034xX, 2018-01-03:
Hey. I've downloaded python, and Nobe.JS. Then fluffd, i've tried everything, and nothing is orking, it just gives me an error.

Error: syntax error
Source: Microsoft JScript compliation error

What am i doing wrong?

XxDJPartyGirl1034xX, 2018-01-04:
Oh, oh! Wait a sec! I clicked on something that opened google chrome, it says "index", but then, it opens the request generator i saw in your vid, but it won't work, it says "no parameters!" and on the top, it says "No command loaded". I turned my Furby on, and nothing happened. Does fluffd open on google chrome if this is the case? plz reply your answers to both of my comments. If you will reply again, i highly  appreciate your help! Thx so much!

pdjstone, 2018-01-05:
If you have access to an Android or Chromebook device you could give my Web Bluetooth project a go: https://github.com/pdjstone/furby-web-bluetooth

Bartman236, 2018-01-18:
Hi pdjstone,

Works perfectly in Chrome on Mac!

XxDJPartyGirl1034xX, 2018-01-22:
I don't have any apple computers... i have an HP computer, and windows 10... i've installed node.js, noble, all the dependencies. Nothing works, i just get a "Syntax error". Does it require Winston...? Plz someone help, i REALLY REALLY want to get this this thing working!

Jeija, 2018-01-23:
I'd like to be able to help you @XxDJPartyGirl1034xX, but it is really difficult to do that over the internet. Maybe you could try to ask a relative, friend or classmate that is even more tech-savvy than you are? Also, there are sometimes people on Silvolf's discord chat server that are very helpful, maybe they could assist you? When helping Silvolf get fluffd running, we had a discrod phone conversation which made communication much easier.

Otherwise, please post screenshots of the exact problem you are running into and the steps you took before, I just can't help you otherwise.
And yes, fluffd requires winston, you need to execute `npm install` in a terminal while you're in the `fluffd` directory to install all of the dependencies.

ja450n, 2018-01-24:
This was brought up on another bug, so I'll respond here. I was able to run this on Win7, Win10, and Ubuntu. The Microsoft JScript Compilation error sounds like you're just double clicking on the fluffd.js file.

You need to run "node fluffd.js" from the command prompt to get the actual service up and running, once the fluffd service is running you can then open the index.html file which will give you a user interface to send commands to the fluffd service.

Running it like this can be a little confusing if you haven't dealt with node.js before. Running "node fluffd.js" is essentially setting up a server on your PC that you are then connecting to with the index.html file.

XxDJPartyGirl1034xX, 2018-01-24:
I have an HP computer with Windows 10. Instructions please? Thanks! I love the help!

ja450n, 2018-01-25:
getting Noble to work is really outside the scope of this project, but in my experience you need to have a compatible Bluetooth radio. 

I have an Asus Motherboard in my Win7 PC with onboard Bluetooth 4.0 and it did not work, and my Ubuntu laptop had a pre BT4 radio, so I had to buy a USB Bluetooth adapter to get the noble library working on these computers.

My Win10 laptop has an Intel BT Radio, which is listed as supported
https://github.com/sandeepmistry/node-bluetooth-hci-socket#windows

XxDJPartyGirl1034xX, 2018-01-25:
Well, my grandfather probably won't let me buy a BT adaptor, i might have a flash drive. So, there's no way to get fluffd to work with no flash drive, or BT radio/Adaptor?

ja450n, 2018-01-25:
The Furby Connect uses Bluetooth Low Energy (BLE) which is part of the Bluetooth 4.0 spec, you need to have a device with a BLE capable radio in it to be able to connect to the Furby Connect.

an earlier comment mentioned a Web Bluetooth based project that might be worth a shot:
https://github.com/pdjstone/furby-web-bluetooth

I've used this on my Android phone under Chrome, and a prior comment said it works in Chrome on a Mac. It probably wouldn't hurt to give it a shot on iOS in Chrome (i talked to someone else via discord about iOS and Safari and it did not work).

pdjstone, 2018-01-25:
Web Bluetooth isn't supported by Chrome on iOS unfortunately. Chrome on an Android phone or tablet is your best bet, as long as it's running Android 6.0 or later.

ja450n, 2018-01-25:
@pdjstone have you seen and/or tried David Parks iOS BleBrowser?

https://itunes.apple.com/us/app/webble/id1193531073?ls=1&mt=8
https://github.com/daphtdazz/BleBrowser

XxDJPartyGirl1034xX, 2018-01-27:
I’ve tried everything, nothing works, I’ll send u screenshots of the errors later.

pdjstone, 2018-01-27:
@ja450n I'm aware of BleBrowser. I've not tried it with my Furby project, but it looks like it supports the required bits of the Web Bluetooth APIs

XxDJPartyGirl1034xX, 2018-02-05:
Pdjstone, Wow, your program works like a charm on my Grandpa’s phone! I would use fluffd, but I can’t get it to work. Thx for making the program! Credit to jeija for introducing me to Furby bluetooth hacking! Thx to all u guys do much!