Jeija/bluefluff #27: Injected audio files not playing on Furby

doanho, 2018-01-03:
Hey, Jeija, 

Thank you very much for making such interesting project. 

I have managed to flash and activate custom DLC file, however my injected audio files are not playing. I am converting via the Gadget Audio Batch converter with 16Khz WAV input as described in your project. 

Do you have any idea what can be wrong? What exact procedure did you use when you injected the songs? 

I would appreciate any kind of help. Thanks.

Jeija, 2018-01-03:
What exactly happens? Do the files not play at all, is there silence instead or just weird noises? Does the regular sound play?
Recently, Context Information Security did a much better job at understanding the DLC file structure, maybe these tools here can help you inject your own files: https://github.com/ctxis/Furby

doanho, 2018-01-03:
The files does not play at all, but the regular sounds play. Do you have any ideas?

I will surely check out the tools. 

Thank you very much.

doanho, 2018-01-03:
I have found out that the audio files are not being injected into the DLC file. Here is the 
[util.zip](https://github.com/Jeija/bluefluff/files/1601065/util.zip)

attachment with the inject_binary I am using. Can you please check it? 

Thank you very much.

closed 2018-01-10 by doanho