Jeija/bluefluff #29: Don't go above action 128 for Input 33

ja450n, 2018-01-14:
Not really an issue, but a warning for others who are playing around. Action 129 and up on Input 33,0,0,x will result in Furby getting stuck in a glitch loop. Sleep mask on/off will break them out of it.

XxDJPartyGirl1034xX, 2018-01-21:
Does this work for NRF connect too? The app?

ja450n, 2018-01-21:
The actions can all be triggered from the nRF Connect app, but the numbers need converted from decimal to hex and prefixed with '1300'

so, for example, action 17,0,0,5 (beatboxing) would be 130011000005 (17 becomes 11 when converted to hex)

XxDJPartyGirl1034xX, 2018-01-22:
Are u able to give me the specific number that glitch Furby out plz and thank u? :D

ja450n, 2018-01-23:
I cannot speak for @Jeija , but I purposefully did not directly provide the value needed to crash a Furby connect. My experience has taught me that directly showing someone how to do something potentially dangerous without that person understanding what they are doing is problematic. If you are going to be experimenting with something that could harm your equipment, it may be in your best interest to research a little more so you are more prepared to handle potential problems you may run into.

XxDJPartyGirl1034xX, 2018-01-24:
Meh, whatever, i just wondered if u could, FYI, i'm not trying to kill my Furby. Since u obviously have a perfect Bluefluff master on your PC, what PC do u have? I have an HP PC, and Windows 10. I've installed it, i get a Microsoft JS "Syntax error" from Windows Script Host... i'll keep trying and update u, and Jeija.

XxDJPartyGirl1034xX, 2018-02-04:
HOOOLLLYY SHIT! U were right, man! I tried this on FurBLE and it glitched my Furby Noo-Tah pretty hard, I just simply had to reset him, he works perfectly now. So, he’s ok.

XxDJPartyGirl1034xX, 2018-02-04:
Also, don’t sent 33,0,0,130, sending this will result in the Furby saying one word over and over again, like, “Wah! Wah! Wah!” Until you press the reset button.

ja450n, 2018-02-04:
the other indexes all seem to end with with a "end of list" type of action where there are one or two  actions that repeat at the end of every list (these actions normally also blink the antenna).

for some reason the name list doesn't do this, and anything above 128 causes a glitch, based on the behaviour it seems like triggering these actions results in some sort of invalid out of bounds memory access (hence the repeating or invalid noises and garbled graphics).

i'm not sure what harm would come from letting the furby sit in this state for any period of time.

XxDJPartyGirl1034xX, 2018-02-05:
I’ve seen the glitch triggered by 129 before, and people have been saying that this glitch only occurs when they tap on the furbling when the Furby’s Bluetooth connection is present within the app, so, the app accidentally sends a name action command that’s above 128, causing it to spazz out. So that’s what causes that glitch. If this is the case, why does it only seem to happen to pink and blue Furby connects? I’m pretty sure it’s happened to other colours.

MonsterrTooth, 2024-10-01:
I know I am BEYOND late to ask here, but as I have very little experience and wanna be careful

Is it possible to straight up brick a Furby irreparably by sending random commands, or am I good?
Since there is no documentation on the specific DLC codes, and I'd love to just try and see what the full spectrum of actions is in general :D

Jeija, 2024-10-01:
I don't think anyone can tell you for sure. Worst thing that happened to me was that I needed to re-insert the batteries, but I didn't ever brick my Furby.

jayasafunctionofe, 2024-10-01:
> I know I am BEYOND late to ask here, but as I have very little experience and wanna be careful
>
> Is it possible to straight up brick a Furby irreparably by sending random commands, or am I good? [...]

I do not work for hasbro, so i can't make any guarantees, but if all you're doing is sending interaction commands you're probably safe, the worse i've seen is a crash that required a reboot.

i have not seen enough research done on the DLC files and how they're loaded to know for sure how unlikely or impossible it would be that you could, in theory, create a DLC file that result in a furby stuck in an unrecoverable state on boot up.  

i would personally hope the "hard reset" would recover from this, but it's hard to say if hasbro planned for people sending unexpected (i.e. corrupt) files to a furby.

as for the "full spectrum of actions", i spend quite a bit of time documenting all the actions i could find here
https://github.com/Jeija/bluefluff/blob/master/doc/actionlist.md

i'm doubtful, but would definitely love to know if, you find any new ones!