Jeija/bluefluff #32: Possible list of all current DLC files

ja450n, 2018-02-08:
Assuming the DLC files all end in zero, this appears to be a list of all current DLC files, can't help but wonder if there is a pattern:

- tu002350.dlc
- tu002560.dlc
- tu002770.dlc
- tu002880.dlc
- tu003000.dlc
- tu003150.dlc
- tu003410.dlc
- tu003550.dlc
- tu010040.dlc
- tu010110.dlc
- tu010160.dlc
- tu010390.dlc
- tu010530.dlc
- tu010600.dlc
- tu010950.dlc
- tu011010.dlc
- tu011090.dlc
- tu011100.dlc
- tu011840.dlc
- tu011880.dlc
- tu011920.dlc
- tu011940.dlc
- tu012640.dlc
- tu012700.dlc
- tu012720.dlc
- tu012750.dlc
- tu012760.dlc
- tu012770.dlc

jeffrafter, 2018-06-19:
* tu020730.dlc
* tr020740.dlc
* tu021270.dlc
* tr021360.dlc

jeffrafter, 2018-06-19:
Russian:

bulkyacronym, 2018-06-20:
Dumb question maybe, but would Russian DLC work on a furby purchased in america?

CylMaddhatta, 2023-04-07:
@ja450n @jeffrafter Sorry for necroing this but do either of you happen to have those DLC files you listed? Hasbro killed their AWS server and I don't know of any archives for these DLC files.

jeffrafter, 2023-04-07:
Oh my... I have a set that is out of date somewhere. I don't think I have all of them at all though...

ja450n, 2023-04-08:
@CylMaddhatta , i've pushed what I managed to scrape up to my fork

https://github.com/ja450n/bluefluff/tree/master/connectworld

@jeffrafter , if you have any files not in my repo let me know and we can try to get it as complete as we can.