Jeija/bluefluff #33: Fluffd finds furby and connects, then fails

marijullama, 2018-03-17:
Hello everyone, first time post here. I am having an issue, fluffd seems to find the furby and connect but then fails. I have the following js error when attempting to execute:

~/Furby/bluefluff-master/fluffd$ sudo node fluffd.js
info: Discovered Furby: da8db8039617
/home/llama/Furby/bluefluff-master/fluffd/fluffcon.js:234
		let characteristicUUIDs = Object.values(FURBY.CHARACTERISTIC);
		                                 ^

TypeError: Object.values is not a function
    at /home/llama/Furby/bluefluff-master/fluffd/fluffcon.js:234:36
    at Peripheral.<anonymous> (/home/llama/Furby/bluefluff-master/fluffd/node_modules/noble/lib/peripheral.js:38:7)
    at Peripheral.g (events.js:292:16)
    at emitOne (events.js:96:13)
    at Peripheral.emit (events.js:188:7)
    at Noble.onConnect (/home/llama/Furby/bluefluff-master/fluffd/node_modules/noble/lib/noble.js:187:16)
    at emitTwo (events.js:106:13)
    at NobleBindings.emit (events.js:191:7)
    at NobleBindings.onLeConnComplete (/home/llama/Furby/bluefluff-master/fluffd/node_modules/noble/lib/hci-socket/bindings.js:223:8)
    at emitMany (events.js:127:13)

Jeija, 2018-03-18:
You are most likely running an older node.js version. You will need node.js version 7.0 or later in order to run fluffd. Unfortunately, Ubuntu ships somewhat outdated node.js packages, you probably got version 6 from the repositories.

marijullama, 2018-03-19:
ah okay! I'll try to get 7.0 up when I get home later this evening and report back how it works. I appreciate the tip thank you!

marijullama, 2018-03-22:
I found my problem, 7.x is EOL and no longer supported on the version of Ubuntu I am running. Looks like I'll have to go to an older distro, or try to compile with 8.x.... :(

marijullama, 2018-03-25:
Alright I installed Mint as I could get nodejs 7.x on it. Got that installed, but when I Run npm install, it throws all sorts of errors :( It seems it does not want to install some dependencies but I am unsure why.