Jeija/bluefluff #36: Alternate power for Furbies

jeffrafter, 2018-04-27:
First, thank you! I have had so much fun using everything in here! 

In a comment here: https://github.com/Jeija/bluefluff/issues/5#issuecomment-275459000 you mentioned:

> Use fresh batteries or a lab power supply, don't use rechargeables!

I'm curious how you powered it and if you used a lab power supply.

sSpeaker, 2018-04-27:
You can power it with 5V power supply or USB.  I'm using simple usb charger from chromecast and everything works well.  If you need more info I can post a photo later.

jeffrafter, 2018-04-27:
Ah, that was what I was hoping 💃 did you open it up and connect in there or did you connect the charger into the battery holder?

sSpeaker, 2018-04-29:
Here is the picture: https://ibb.co/hWZytc
Black board is just for micro USB connector. 
Black wire - minus. You don't need to solder wires, just push them under the battery contacts.  Also there is small cut near tail, for usb cable.