Jeija/bluefluff #43: Need help getting the audio files on the Furby

SaSoGaming, 2019-01-01:
Hello. I've been recently tinkering around with my Furby and after lots of trial and error, I have been able to get the fluffd client working properly and I've been able to send specific actions, antenna colors, etc, to the Furby. I've been wanting to now add my own audio files and possibly images to the Furby, but decided to start with the audio first because it seemed much easier. Through lots of research and forum searching, I've been able to create an a18 audio file with the specific bitrate and such. I also have the tu003410.dlc file and all the Python programs for the audio. But now, I am unsure of where to go from here. I believe that I have correctly edited the inject_binary script to target the specific files but I am unsure if it worked and what to do with it using the fluffd client. Is there anyone that could give me some more specifics on what to do next?

I've supplied my inject_binary script below (and changed my real device name to just "name").

[Furby Audio Example.zip](https://github.com/Jeija/bluefluff/files/2719626/Furby.Audio.Example.zip)

SaSoGaming, 2019-01-02:
I believe I have successfully created my DLC file for Furby, but I am running into one final issue. As stated in a closed issue, I am currently getting this error message:

verbose: Sending flashdlc command to all Furbies, params: filename=TU003410.DLC, dlcfile=./Music_Files/tu003410.dlc
verbose: generalPlusWrite: 5000113dd60254553030333431302e444c430000
verbose: GP notification: 2402
info: FlashDLC: Got Ready to Receive
info: FlashDLC: Got Ready to Receive
info: FlashDLC: Got Ready to Receive
verbose: generalPlusWrite: 00
verbose: GP notification: 2403
verbose: generalPlusWrite: 00

Then it continues producing 0's. My path correctly leads to the folder and I have tried using both my edited DLC file and the default DLC but to no avail. In the closed issue, you said "the user fluffd is running as has read permissions". What are these permissions and where can I find them? I feel like I am close to flashing the files, so is there any other tips you could give me on how to fix this error?

SaSoGaming, 2019-01-02:
[DLC Files.zip](https://github.com/Jeija/bluefluff/files/2721875/DLC.Files.zip)

Here's the files I'm working with. The first one is a completely unedited file that is the default DLC file. The second is edited and has a longer song around 3 and a half minutes with the action command being 75.0.4.0. The third is also edited but is shorter at 1 and a half minutes with the action command at 75.0.0.3. The error is happening to all of the DLC files and it seems like the transmission is getting corrupted or glitched somehow. When I restart both my Furby and the server after flashing the DLC, I load up the debug screen on Furby to see that four of the 1 MB DLC slots are filled and activated without me activating them. They still do not play any of the actions/songs I have put on the file. I will continue to try different options and tinker around to find out what's failing the transmission. Thank you for responding!

Jeija, 2019-01-13:
Sorry for the late reply, I've been kind of busy... I can tell you that it is certainly not your DLC file's fault. I have flashed `tu003410-portal.dlc` to my Furby and it seems to work fine, see https://www.youtube.com/watch?v=Lb_zsFIHLP8

Maybe your BLE hardware causes some problems? Have you tried using fluffd on a different computer or with a different BLE USB stick?

ollieisnumber09-netizen, 2026-08-20:
i realsie its been atleast 7 years but does anyone know how to convert a wav file into dlc?

BNNorman, 2026-08-20:
It's been a long time since I did this - but I could not get my DLC to upload. Anyway, here are my brief notes on how to convert WAV to A18 for inclusion in the DLC:-

#1 install rust
#2 pip install a1800_codec  # python! This is compiled by rust
#3 Build the a1800_codec using rust then add path ./target/release to PATH to your environment (I forget this bit)
#4 run: a1800_codec encode <?>.wav  <?>.a18 

WAV must be 16khz mono. Use audacity to convert if needed

Sorry if it's a bit hazy but might help.

ollieisnumber09-netizen, 2026-09-04:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Thanks