Jeija/bluefluff #45: BlueotoothHCI dont compile in linux.

Neocop, 2019-07-22:
Hi Jeija. When I execute npm install, I have a compilation error in BluetoothHciSocket. 

gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:190:23)
gyp ERR! stack     at ChildProcess.emit (events.js:193:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
gyp ERR! System Linux 5.2.1-arch1-1-ARCH
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/neocop/src/bluefluff/fluffd/node_modules/bluetooth-hci-socket
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v5.0.2
gyp ERR! not ok 
-----------

I'm using archlinux x86_64. Thank you

Jeija, 2019-07-22:
This problem is related to https://github.com/noble/node-bluetooth-hci-socket/issues/84, it's not bluefluff's fault. As a workaround, you can install the LTS version of nodejs (`pacman -S nodejs-lts-carbon` on Archlinux). Please let me know if this helps / close the issue if it does.

Neocop, 2019-07-22:
Ok Jeija, Problem solved. Now I can open the html interface and I can see the options at the left.
Thank you for your work.  :)

closed 2019-07-22 by Jeija