Jeija/bluefluff #49: Original DLC files distributed by Hasbro

sanjanalreddy, 2019-10-29:
Hi, I got a furby connect recently so I don't have any of the DLC files which were distributed by Hasbro through the world connect app. The app doesn't let me access these files anymore. Is there a way to get these files somehow? Is it already on the github repo somewhere?

Jeija, 2019-10-29:
Hi @sanjanalreddy!
You can just download the DLC files from Hasbro's server at Amazon AWS.
The URL you need to use is
```
http://chromecastpd.s3.amazonaws.com/furbyconnect/cinemav010/<DLC FILE NAME>
```
where you need to replace `<DLC FILE NAME>` with the actual filename of the DLC file, so for instance
```
http://chromecastpd.s3.amazonaws.com/furbyconnect/cinemav010/tu013530.dlc
http://chromecastpd.s3.amazonaws.com/furbyconnect/cinemav010/tu011880.dlc
http://chromecastpd.s3.amazonaws.com/furbyconnect/cinemav010/masterlist.xml
```

Thanks to ja450n and jeffrafter, there is a (hopefully complete) list of DLC filenames available from the server, see #32 . I hope this helps you!

sanjanalreddy, 2019-10-29:
Thank you! I will try this!

closed 2019-10-29 by sanjanalreddy