Jeija/bluefluff #50: Issue building on Ubuntu 19.10

RothNath, 2019-12-20:
Tried to paste in the relevant sections with errors while anonymizing as possible. Thank you again for your work on this project. Really enjoyable modifying Furbies <3. 

python make g++ libudev-dev are all installed. 

Not sure what to try given these errors. 

../src/BluetoothHciSocket.cpp: In static member function ‘static void BluetoothHciSocket::Init(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
../src/BluetoothHciSocket.cpp:128:82: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
  128 | luetoothHciSocket").ToLocalChecked(), tmpl->GetFunction());
      |                                                         ^

bluefluff-master/fluffd/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:305:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 5.3.0-24-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /bluefluff-master/fluffd/node_modules/bluetooth-hci-socket
gyp ERR! node -v v13.5.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN fluffd@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

vvegastl, 2020-01-20:
The problem I universally ran into whether trying to install this either on Linux (bluetooth-hci-socket) or Mac (xpc-connection) was dependancies having been abandoned by their maintainer, thus not working with newer versions of node, etc. In some cases, someone might have forked one those dependencies and got it working, but what I ultimately did was downgrade to Node 8 and then I was able to successfully install.

Zebravado, 2020-03-24:
So I have been testing this myself. bluetooth-hci-socket has been abandoned. There is a replacement for it with bluez in node. I will open a separate issue requesting help locating where the flippity flop the gosh darn call for that library is so that I can just replace it.