Jeija/bluefluff #55: Need help with disassembly

Tusskie, 2020-11-12:
I'm trying to turn my connect into a walking smart home hub abomination. I need carefully taking the poor thing apart and getting inside to change a few things. Using this tutorial in case you're wondering
https://howchoo.com/g/otewzwmwnzb/amazon-echo-furby-using-raspberry-pi-furlexa

Jeija, 2020-11-13:
Hi @Tusskie! The tutorial you linked to is about the "original" Furby, not the Furby Connect. For taking apart Furby Connect, I suggest you take a look at this: https://www.youtube.com/watch?v=JlUDiag7i6c

Please note that turning Furby Connect into Furlexa is going to be significantly more difficult if you want to keep the eye LCDs; with Furby Connect, you can just connect the eye motors to some GPIO pins, but with Furby Connect you'd probably have to install new LCDs for the eyes to make it work. Here's a project where someone programmed custom eye animations for their Furby: https://hackaday.io/project/9386-open-furby, see also https://www.youtube.com/watch?v=u3GzCX6JvRM

closed 2020-11-13 by Jeija

Tusskie, 2020-11-14:
Thank you. I just didn't want to damage any motors needed. I have a replacement pi if necessary :)