Jeija/bluefluff #56: Fluffd crashes on debug

tluxton, 2021-04-03:
I can connect to fluffd.js fine and run commands through Preprogrammed Actions in fluffd-client to Furby fine. But when I click Send button on 'Debug Screen' I get an error pop up in browser 'This page says: error: TypeError: callback is not a function'. And in command prompt I see the same error but then message: 'Unknown Peripheral edd5da2184e2'  several times before getting the line: 'MaxlistenersExceededwarning: Possible Eventemitter memory leak detected. 11 write listeners added  to [characteristic]. Use emitter.setMaxListeners() to increase limit. Meahwhile Furby mutter some crap about "Why so Soon" and shuts down. 

Would appreciate a step in the right direction as this has taken ages to get to this stage! I'm guessing Im doing something stupid with not using debug mode correctly?

Cheers
 
[image]

[image]
[image]

Jeija, 2021-04-06:
Iit seems like you found a bug in the code. I have pushed a fix to this repository (https://github.com/Jeija/bluefluff/commit/42d1338037c645ccef0851b82f72a4180e2cbaee), could you update your local files (or manually edit that single line in fluffaction.js) and see if it works now?

I haven't actually tested the fix with my Furby yet, so let me know in case it doesn't work for you.

tluxton, 2021-04-10:
awesome thanks it works now! just another question in fluffd-client what format should the Delete DLC command be in? ie something like 75.0.2.6 ?

Jeija, 2021-04-10:
No, it is the DLC slot number, so something like "0" or "1" or "2" or "3"..., depending on which slot you flashed a DLC to.
The debug eyes provide a hint as to which DLC slots are in use, see https://github.com/Jeija/bluefluff/blob/master/doc/flashdlc.md.

closed 2021-04-10 by Jeija