Jeija/bluefluff #57: No compatible USB Bluetooth 4.0 device found!

ChR1sMC, 2021-04-12:
After executing node fluffd.js I get the error: 

```
C:\Users\...\bluefluff-master\fluffd\node_modules\bluetooth-hci-socket\lib\usb.js:70
    throw new Error('No compatible USB Bluetooth 4.0 device found!');
    ^

Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (C:\Users\...\bluefluff-master\fluffd\node_modules\bluetooth-hci-socket\lib\usb.js:70:11)
    at BluetoothHciSocket.bindRaw (C:\Users\...\bluefluff-master\fluffd\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (C:\Users\...\bluefluff-master\fluffd\node_modules\noble\lib\hci-socket\hci.js:101:35)
    at NobleBindings.init (C:\Users\...\bluefluff-master\fluffd\node_modules\noble\lib\hci-socket\bindings.js:82:13)
    at Noble.<anonymous> (C:\Users\...\bluefluff-master\fluffd\node_modules\noble\lib\noble.js:57:24)
    at processTicksAndRejections (internal/process/task_queues.js:75:11)
```
I tried it on Lenovo ThinkPad T580 also on a macbook air 2014 and I get the same error. Both devices seem to have a compatible Bluetooth 4.0. Is it mandatory to have an USB Bluetooth ?

Jeija, 2021-04-12:
fluffd relies on [noble](https://github.com/noble/noble) for Bluetooth Low Energy.
The error you are seeing was not generated by fluffd itself, but [node-bluetooth-hci-socket](https://github.com/noble/node-bluetooth-hci-socket). I don't know which bluetooth devices exactly are supported, but maybe https://github.com/noble/node-bluetooth-hci-socket/issues/118 or https://github.com/noble/noble/issues/472 can help?

If not, you could also try https://github.com/pdjstone/furby-web-bluetooth.

ChR1sMC, 2021-04-14:
> fluffd relies on [noble](https://github.com/noble/noble) for Bluetooth Low Energy.
> The error you are seeing was not generated by fluffd itself, but [node-bluetooth-hci-socket](https://github.com/noble/node-bluetooth-hci-socket). I don't know which bluetooth devices exactly are supported, but maybe [noble/node-bluetooth-hci-socket#118](https://github.com/noble/node-bluetooth-hci-socket/issues/118) or [noble/noble#472](https://github.com/noble/noble/issues/472) can help?
> 
> If not, you could also try https://github.com/pdjstone/furby-web-bluetooth.

Yes, that helped me, thanks a lot!

closed 2021-04-14 by ChR1sMC