Jeija/bluefluff #61: Still working?

Noachh, 2022-01-22:
Hey! I was wondering if this still works. I know it's an old project but before i start wasting my time again trying to upload my own audio file and failing again i wanted to know if it's still at all possible. Thanks!

Jeija, 2022-01-23:
It is possible to get this code working, but it is not compatible with the latest node.js version. Also, [noble](https://github.com/noble/noble) appears to be somewhat discontinued it seems? Any help in updating this project is appreaciated, but unfortunately, I don't have the time for it.

Noachh, 2022-01-23:
I wish i could help but  don't know anything about coding (planning to learn in the future). I wish someone would update this who has the knowledge and time to do so at least. Well thanks for the quick reponse tho! I really appreciate it :)

CylMaddhatta, 2022-01-27:
Noachh, what OS are you using? I'm currently trying to get this working so if I have any success I'll let you know.

Noachh, 2022-01-27:
Hey CylMaddhatta, i use both Kali Linux and Windows 11, so either will work. Good luck with it! I hope you can get it to work.

CylMaddhatta, 2022-01-27:
Good news Noachh, I got it to work!

I'm on Mac OS 11.6

There are a couple of things you need to do in addition to following the normal instructions.

1). I did this on the latest version of Node.js (16.13.2) so install that.

2). Noble is no longer being developed. It has been picked up by @abandonware/noble. Run the following

`Install with npm i @abandonware/noble`

3) Navigate to the fluffd folder in your command line and run `npm rebuild`

4). As laid out by @hartos in [https://github.com/Jeija/bluefluff/issues/20](url)
Changes:

file::package.json

in the section `"dependencies": {`
change
`"noble": "1.9.1",`
to
`"@abandonware/noble": "^1.9.2-2",`
and
Also change `"winston": "3.1.0",`
to
`"winston": "3.1.0"`
Because JS will yell at you about syntax if you don't remove the comma

in
file::fluffd.js
change
`const noble = require("noble");`
to
`const noble = require("@abandonware/noble");`

Your Furby will automatically connect. The UI is browser based using the index.htlm file.

I have not attempted uploading flashed DLC yet. I plan to do a full 2022 install guide once I get this working.

Jeija, 2022-01-27:
@CylMaddhatta That is great to hear 👍 ! Feel free to open a pull request to get your changes upstream if you'd like to!

Noachh, 2022-01-28:
Awesome work! Really appreciate the work you're doing :+1: 

Yeah i always wanted to upload my own DLC but i never succeeded, hope you can find a way to make it work! I'm gonna try the things u said above.

Noachh, 2022-01-28:
So i just tried uploading my own audio file into the furby with a lot of trial and error. As you can see in the image i got it acivated but when i tried action 75.0.2.6 no sound came out. Only movements, maybe when you try you could find out why this happens? Because i got no clue what's wrong if i'm honest haha

[image]

CylMaddhatta, 2022-01-28:
You got yours to flash?

I'm getting 2405 notifications at the start of the upload. It's dropping connection as well.

Fortunately I found that the FurBLE project works for flashing but I'm not sure how to upload custom DLC to that yet. Was just about to start seeing if I could use that as a work around.

Noachh, 2022-01-28:
I've tried using FurBLE as work around too, i think you need to start a local server with the files from FurBLE and then add your own dlc in the list to choose from, idk how to explain it better. I'll see if i can do it tommorow again it's been a while since i did it

CylMaddhatta, 2022-01-28:
I spun up a discord server to scream into: https://discord.gg/FhzxgeH4

MikeDBloke, 2022-02-09:
Still stuck on this. After following what @CylMaddhatta posted and doing what the main page says
Im on Ubuntu
~$ node fluffd.js
node:internal/modules/cjs/loader:936
  throw err;
  ^

Error: Cannot find module '/home/dbloke/fluffd.js'
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:81:12)
    at node:internal/main/run_main_module:17:47 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}
The fluffd.js isnt in the main folder but is in another one where it was installed to /fluffd/fluffd, 
Im assuming its not finding something, I moved to teh folder with the js file and now I get

~/bluefluff/fluffd$ node fluffd.js
/home/dbloke/node_modules/@abandonware/noble/lib/hci-socket/hci.js:121
      this._socket.bindRaw(this._deviceId);
                   ^

Error: ENODEV, No such device
    at Hci.init (/home/dbloke/node_modules/@abandonware/noble/lib/hci-socket/hci.js:121:20)
    at NobleBindings.init (/home/dbloke/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:93:13)
    at /home/dbloke/node_modules/@abandonware/noble/lib/noble.js:61:24
    at processTicksAndRejections (node:internal/process/task_queues:78:11) {
  errno: 19,
  code: 'ENODEV',
  syscall: 'bind'
}

CylMaddhatta, 2022-02-09:
@MikeDBloke seems like @Abandonware/noble isn't happy. I updated the discord link in case it was broken. I have a write up pinned for how to get the DLC loaded.

MikeDBloke, 2022-02-10:
@CylMaddhatta Cheers ill give it a look when I next feel like fiddling with it