Jeija/bluefluff #63: May-doo won't connect!

DeveloperWOW64, 2022-04-23:
[image]
From what I can understand the application is defaulting to the Bluetooth adapter; but I don't have one.
I installed everything according to [CylMaddhatta](https://github.com/Jeija/bluefluff/issues/61#issuecomment-1023607268)..
I tried everything with PNPM as well as NPM; still didn't work. Tried in PS, CMD.
What can I do?

CaelansStuff, 2022-04-25:
I would recommend you talk about it in CylMaddhatta's Furby hacking discord server https://discord.gg/ta33H7PjRb

Tom-Neverwinter, 2024-08-21:
> CylMaddhatta's Furby hacking discord

invite link is dead sadly.

Lamparter, 2024-08-21:
> > CylMaddhatta's Furby hacking discord
> 
> invite link is dead sadly.

Is it? https://discord.com/invite/qcqQSxhj