Jeija/bluefluff #67: Where should i go from "node fluffd.js"

psychicspectre, 2024-11-07:
When i try use the above command, nothing happens? 
[image]
I cant type in the terminal after i send the command, does anyone have any ideas? Thanks. :)

Jeija, 2024-11-07:
Just follow [these instructions](https://github.com/Jeija/bluefluff?tab=readme-ov-file#sending-commands-using-fluffd-client):

> You can now open `fluffd-client/index.html` in your favorite web browser to take control of Furby Connect. Most menu items should be self-explanatory. See [the DLC flashing documentation](doc/flashdlc.md) for information on DLC-related commands

So, using your file manager, open the `fluffd-client` folder and open `index.html` in your web browser.

psychicspectre, 2024-11-08:
I will try that and get back to you, thank you so much!

closed 2024-11-09 by psychicspectre